pcb2gcode/pcb2gcodeGUI #1: it dont complile

kagouraki, 2015-04-26:
i am trying to compile this an i run all the steps in the readme file. In terminal if i type the make command i am getting the flowing error:

kagouraki@ubuntu:~/pcb2gcodeGUI$ make
/usr/lib/i386-linux-gnu/qt5/bin/uic mainwindow.ui -o ui_mainwindow.h
/usr/lib/i386-linux-gnu/qt5/bin/uic outputwindow.ui -o ui_outputwindow.h
g++ -c -pipe -O2 -std=c++0x -Wall -W -D_REENTRANT -fPIE -DQT_NO_DEBUG -DQT_WIDGETS_LIB -DQT_GUI_LIB -DQT_CORE_LIB -I/usr/share/qt5/mkspecs/linux-g++ -I. -I/usr/include/qt5 -I/usr/include/qt5/QtWidgets -I/usr/include/qt5/QtGui -I/usr/include/qt5/QtCore -I. -I. -o main.o main.cpp
In file included from mainwindow.h:36:0,
                 from main.cpp:20:
argaction.h: In member function ‘void argAction::insert(QString, QDoubleSpinBox*)’:
argaction.h:80:45: error: expected primary-expression before ‘)’ token
argaction.h: In member function ‘void argAction::insert(QString, QSpinBox*)’:
argaction.h:86:45: error: expected primary-expression before ‘)’ token
argaction.h: In member function ‘void argAction::insert(QString, QCheckBox*)’:
argaction.h:92:45: error: expected primary-expression before ‘)’ token
argaction.h: In member function ‘void argAction::insert(QString, QLineEdit*)’:
argaction.h:98:45: error: expected primary-expression before ‘)’ token
argaction.h: In member function ‘void argAction::insert(QString, QComboBox*)’:
argaction.h:104:45: error: expected primary-expression before ‘)’ token
argaction.h: In member function ‘void argAction::insert(QString, QWidgetPair<QRadioButton, QRadioButton>*)’:
argaction.h:110:45: error: expected primary-expression before ‘)’ token
make: *** [main.o] Error 1

Corna, 2015-04-27:
It is probably related to the C++11 requirement
The latest version does not require C++11 anymore, can you please test it?
By the way, what version of gcc are you using? Which distro?

kagouraki, 2015-04-27:
ok i made it. maybe the git clone https://github.com/pcb2gcode/pcb2gcodeGUI.git command dont downloads the latest version. I downloaded the hole project zip, extracted and now is working.

Thank you

Corna, 2015-04-28:
I've released the fix after your bug report
If you have cloned a repository you can download the latest modifications (without downloading the whole project again) going in the project directory and launching a git pull, then recompile the project with a qmake && make

closed 2015-04-28 by Corna