pcb2gcode/pcb2gcodeGUI #3: pcb2gcodegui not finding pcb2gcode

ericgunstone, 2016-09-07:
Hi, hope someone can help with this..
I cannot get pcb2gcodeGUI to work correctly, it does start up but after I insert the details and press start I get this as output...
Starting pcb2gcode --back=/home/eric/Kicad/Stepper/gerber/Stepper-F.Cu.gtl
--output-dir=/home/eric/Kicad/Stepper/gerber --dpi=1000 --metric=true --metricoutput=true
--mirror-absolute=false --optimise=true --tile-x=1 --tile-y=1 --zchange=10.0000 --zero-start=true
--zsafe=5.0000 --extra-passes=0 --mill-feed=600 --mill-speed=10000 --offset=0.1000 --zwork=-0.0500
 
 Error: You've supplied an invalid parameter.
 Note that spindle speeds are integers!
 Details: unrecognised option '--noconfigfile'

 can you help explain what is happening here and how to fix it, I don't know where any config files
are in either pcbsgcode directory or pcb2gcodeGUI directory

this is on debian jessie 8.5 with KDE
Also, how does the origin get set to match the cnc machines co-ordinates?
any help is appreciated
  Eric

corna, 2016-09-08:
I think you're using an old pcb2gcode version. Have you downloaded pcb2gcode from the Debian repo?
Run `./pcb2gcode --version` or click 'Help -> About pcb2gcode' in pcb2gcodeGUI and tell me the version

corna, 2016-09-08:
Duplicate of pcb2gcode/pcb2gcode#50

closed 2016-09-08 by corna