pcb2gcode/pcb2gcodeGUI #4: Create 2 sided PCB

ericgunstone, 2016-09-11:
Hi,
after getting pcb2gcodeGUI working I'm now trying to work out how to do 2 sided pcb's by creating some alignment holes then flipping the board around the x-axis so I can put the board down in the same location as the first side, this would require a mirror image of the top I guess.

Can anyone recommend steps to follow to achieve this
Eric

ericgunstone, 2016-09-11:
I have been reading the pcb2gcode notes and playing around with linuxcnc simulator and I realize my question is almost irrelevant as I see how pc2gcode does it, however I still don't know how to maintain registration, so I can know the flipped board will still have the identical o,o start point and be aligned correctly.

Sorry if I seem a slow learner.

Eric

corna, 2016-09-12:
Hi,
pcb2gcode mirrors the image internally, you don't have to think about it.

With _mirror-absolute=false_ (the default) the back X-coordinates are mirrored along the board center and, after the milling, drilling and cutting of the front, you have to flip the board, like this

     A --- D       D --- A
     |     |  -->  |     |
     B --- C       C --- B 

placing the board in the **same place** as before.
With _zero-start=true_ both the front and the back will start at (0,0)

You can also manually add alignment holes in your PCB: if you place them at the same distance from the board edges, you can re-align them after the flipping

     +---------+
     | O     O |
     |         |
     | O     O |
     +---------+

Note that the default workflow for a double-sided PCB is

     mill front --> drill front --> cut front --> FLIP BOARD --> mill back

You can change this with the parameters _drill-side_ and _cut_side_.

The new core (activated with _vectorial=true_) has not been tested yet with double-sided PCBs, it would be really appreciated if you could test it and report the results.

> Sorry if I seem a slow learner.

It's not you, the documentation is very poor ;)

Is it clear now?

ericgunstone, 2016-09-13:
Thanks, yes that's a lot clearer now. , I will have a go at it in the next few days and try this vectorial=true as well and let you know.

Eric

closed 2016-09-13 by ericgunstone