pcb2gcode/pcb2gcodeGUI #5: Compilation issue

kosme, 2016-09-27:
Just pulled the latest commits and I get the following error when building:

In file included from mainwindow.h:39:0,
                 from main.cpp:20:
argaction.h: In member function ‘bool argAction::setValue(QString, QString)’:
argaction.h:92:9: warning: ‘auto’ changes meaning in C++11; please remove it [-Wc++0x-compat]
         auto pos = objects.find(key);
         ^
argaction.h:92:14: error: ‘pos’ does not name a type
         auto pos = objects.find(key);
              ^
argaction.h:94:13: error: ‘pos’ was not declared in this scope
         if (pos != objects.end())
             ^
argaction.h: At global scope:
argaction.h:90:17: warning: unused parameter ‘key’ [-Wunused-parameter]
     inline bool setValue(const QString key, const QString value)
                 ^
make: *** [main.o] Error 1

corna, 2016-09-27:
Try to add

     CONFIG += c++11

at the end of pcb2gcodeGUI.pro, re-run qmake and rebuild

kosme, 2016-09-28:
That fixes it but now shows a new error.
In file included from mainwindow.cpp:21:0:
ui_mainwindow.h: In member function ‘void Ui_MainWindow::setupUi(QMainWindow*)’:
ui_mainwindow.h:279:21: error: ‘class QTabWidget’ has no member named ‘setTabBarAutoHide’
         optionsTab->setTabBarAutoHide(false);
                     ^
make: *** [mainwindow.o] Error 1

corna, 2016-09-28:
With commit 280cf3a1c29b794ad68bfd9426c3132e10eccee0 I've removed the C++11 code, and with 832e260d0e81afe09fab98e2e3e1625bd6c5a8ab I've removed the ui properties unsupported by Qt 5.0.0
Can you pull the latest changes and try again?

kosme, 2016-09-28:
It now compiles nicely. The only thing, really small detail, is that now the GUI version is ahead of the CLI version.

closed 2016-09-28 by kosme

corna, 2016-09-28:
I know, I'm going to release the new CLI version very soon