pcb2gcode/pcb2gcodeGUI #11: pcb2gcode unrecognised option

proffalken, 2018-05-16:
Hello,

I'm running Arch Linux with the latest versions of PCB2Gcode and PCB2GCodeGUI from the AUR (https://aur.archlinux.org/packages/pcb2gcode/)

When I run the GUI, as soon as I try to import my files or preview what's going on, I get the following error:

```
Error while processing input files (error code 101)

pcb2gcode output:
Error: You've supplied an invalid parameter.
Note that spindle speeds are integers!
Details: unrecognised option '--zchange-absolute=true'
```

If I remove this value from the command line, the gcode appears to generate properly, however it would be good to be able to preview in the UI first.

eyal0, 2018-05-17:
Seems like the only way that this happens is if the pcb2gcode version and the pcb2gcodegui version are out of sync.  In pcb2gcodegui, under Help, you can pull up the versions of each.  What are they?

proffalken, 2018-05-18:
PCB2gcode: 1.3.2
PCB2GcodeGUI: 1.3.2-

proffalken, 2018-06-08:
@eyal0 - I've just had a chance to come back to this, and when I'm running `pcb2gcode 1.3.2` and pcb2gcodeGUI 1.3.2-1` then I get this error message.

I have now recompiled the latest versions from source, and I still get the error.

Is there anything else I can try?

eyal0, 2018-06-10:
Is it possible that you have two versions of pcb2gcode installed and pcb2gcodeGUI is running the wrong one?

When you type pcb2gcode --help, is the zchange-absolute option listed?

eyal0, 2018-06-18:
> Perfect, thanks.
>
> The source version was in /usr/local/bin with the correct flags, and the
> "wrong" binary was overriding this in /usr/bin with the wrong flags.
>
> The only error I get now is that mirror-absolute must be set to true as
> it is deprecated, but that's easily fixed by updating the appropriate tick
> box in the UI.
>
> Thanks for your help.

Great! I guessed at the bug and got lucky.

mirror-absolute false was kind of useless because you could never really
tell where the mirroring would occur.