pcb2gcode/pcb2gcodeGUI #12: cluttered interface on HiDPI screen

AlexeyGusev, 2018-12-02:
This is how the interface elements look on a screen with 175% scaling:

[image]

- text is too big, does not fit the placeholder. In other words, the placeholders do not scale
- tab pane does not resize, even manually, so some elements stay hidden under the bottom of the pane

eyal0, 2018-12-03:
This project is not very maintained, sorry.  If you have changes you want, you might need to do them yourself.  I'm willing to review code if you need.

AlexeyGusev, 2018-12-03:
>  If you have changes you want, you might need to do them yourself.

ehh, I don't know Qt stuff.

eyal0, 2018-12-03:
Yeah, me neither.