pcb2gcode/pcb2gcodeGUI #17: pcb2gcode unexpectedly crashed.Please report this error to the project manager.

asps946701, 2019-05-14:
Hi 
I first use **pcb2gcodeGUI** on raspberry pi 3B+ with raspbian but when I output gcode it run to crash.
How can I solve this problem ？
Thanks very much

My gerber file [https://drive.google.com/file/d/1GfRjAC3iF0Dh8PJmWjGix1KkjMTl_5CT/view?usp=sharing](url)

**pcb2gcode --version**
2.0.0
Git commit: v2.0.0-102-g6b71101
Boost: 106200
Gerbv: 2.6.1

**when I startup pcb2gcodeGUI , theremin massage :**
libEGL warning: DRI2: failed to authenticate
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.
Gtk-Message: GtkDialog mapped without a transient parent. This is discouraged.

**when I output the gcode, the output window massage :**
Importing front side... DONe.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... DONE.
Exporting back... 
pcb2gcode unexpectedly crashed. Please report this error to the project manager.

eyal0, 2019-05-16:
I don't support pcb2gcodeGUI so I don't know if I will be able to help you.

First let's try to figure out if it's a bug with pcb2gcode or pcb2gcodeGUI.  pcb2gcodeGUI runs pcb2gcode and you can see the command line that it's running.  Want to try running that command line directly on the RPi, without pcb2gcodeGUI, and see what happens?

Also, you can try running the integration tests by running ./integration_tests.py to see if even those work.

asps946701, 2019-05-16:
> I don't support pcb2gcodeGUI so I don't know if I will be able to help you.
>
> First let's try to figure out if it's a bug with pcb2gcode or pcb2gcodeGUI. pcb2gcodeGUI runs pcb2gcode and you can see [...]

Thanks you reply. I will try it tomorrow TT