pcb2gcode/pcb2gcodeGUI #20: New release

akstrfn, 2021-04-01:
Hi, pinned release is very old and does not work due to handling of `--optimise` parameter. Since this is already patched it would be useful to pin new release.

the-snowwhite, 2021-04-04:
Sure can you describe the steps to pin a new release ?

eyal0, 2021-04-04:
I've never used the GUI and it might not be compatible with the latest version of pcb2gcode.

the-snowwhite, 2021-04-07:
I haven't been using pcb2gcode for a while. The last time update is here:
[Update to work with pcb2gcode 2.1.0](https://github.com/pcb2gcode/pcb2gcodeGUI/pull/19)
Now PCB2gcode is v2.3.0. 
Looking at the release notes I can't see any apparent changes to parameters ?

the-snowwhite, 2021-04-07:
https://github.com/pcb2gcode/pcb2gcodeGUI/releases/tag/v1.3.3

akstrfn, 2021-04-07:
As mentioned `--optimise` makes 1.3.2 fail due to change from float to bool and even though the fix was in master branch, 1.3.2 was released in 2016.

@eyal0 I was just testing it for the first time (as well as pcb2gcode) and GUI is just a wrapper around cmd for easy option setting and resulting images viewing. If GUI would just remove commands that are unchecked and leave the default values of pcb2gcode it would basically be almost always "compatible" however it would have less features if new commands are added to pcb2gcode. The way it works now best would be to just generate GUI from millproject file.

Anyway now they both work in archlinux (although I had to patch pcb2gcode):
- https://aur.archlinux.org/packages/pcb2gcodegui/
- https://aur.archlinux.org/packages/pcb2gcode/

closed 2021-04-07 by akstrfn