pcb2gcode/pcb2gcodeGUI #24: can possible remove repeated lines?

ashokavarthanan, 2022-08-11:
hi.
when I using "Extra Passes 10" the result G-code is more repeated lines.
can solve this?

[image]

Holotronic, 2022-08-11:
@ashokavarthanan Pcb2gcodeGUI is just a frontend for pcb2gcode
the parameter is named extra-passes so try going to https://github.com/pcb2gcode/pcb2gcode to get your issue resolved

the-snowwhite, 2022-08-11:
@ashokavarthanan Pcb2gcodeGUI is just a frontend for pcb2gcode
the parameter is named extra-passes so try going to https://github.com/pcb2gcode/pcb2gcode to get your issue resolved

ashokavarthanan, 2022-08-12:
> @ashokavarthanan Pcb2gcodeGUI is just a frontend for pcb2gcode the parameter is named extra-passes so try going to https://github.com/pcb2gcode/pcb2gcode to get your issue resolved

Solved Thanks .
Pcb2gcode command line tool  Solved this.

closed 2022-08-12 by the-snowwhite