pcb2gcode/pcb2gcodeGUI #25: pcb2gcode will crash without reason

Bastler100, 2022-09-04:
Hello,
I use pcb2gcode some time and it works very well. At one time the program always crashes with this messages:
Importing front side... not specified.
Importing back side... DONE.
Importing outline... DONE.
Processing input files... 
This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
terminate called after throwing an instance of 'Glib::FileError'
I use V1.3.2-1 Win64, I use only copper back, board outline and the drill-files.
Many thanks for every answer
Hans

Holotronic, 2022-09-04:
@Bastler100 
You may want to try building a [newer windows version](https://github.com/pcb2gcode/pcb2gcodeGUI#build-on-windows
) with the latest [release](https://github.com/pcb2gcode/pcb2gcodeGUI/releases/tag/v1.3.3
)  ?

If its a bug for pcb2gcode (as you write)please submit your issue here: https://github.com/pcb2gcode/pcb2gcode

eyal0, 2022-09-06:
The latest pcb2gcode ought to compile on Windows but I'm unable to test it because I don't use Windows!

If you know how to compile and you want to try, I'll assist you with any errors.

Bastler100, 2022-09-07:
Hello,
pcb2gcode works now correct. The voronoi modus works in the newer version very well. In the outline gerberfile was a problem
with the boardcontour.

eyal0, 2022-09-07:
I fixed a bug in board contour once where one of the tools was creating paths that were almost but not exactly closed.

Anyway, if it's working, close the bug?

the-snowwhite, 2022-09-07:
@Bastler100 
Im glad that you solved your issue and thanks for following up and reporting back.
@eyal0  Thanks for keeping an eye on this thread
I'll close now

closed 2022-09-07 by the-snowwhite