pcb2gcode/pcb2gcodeGUI #29: multiple "depth" passes for isolation milling

m-anish, 2025-01-08:
Hi. 

I'm using a v bit to mill isolation around traces, and i feel that I need to do multiple depth passes, otherwise I risk breaking the bit. 

For example, the default setting of 0.05mm is just to small for me, and I feel I need to increase it to 0.1 or more to properly mill the isolation around traces. There doesn't seem to be any setting for that?

I can ofcourse manually do it, but wondering if there's an option that I may be missing.

eyal0, 2025-01-08:
Multiple depth for milling! Ah. Hmm, I never had that request because most people are on 1oz copper and it's pretty thin!

What if you just run pcb2gcode twice, at shallow depth and then at deep?

I don't like to add too many flags to the program because it's already got so many!

m-anish, 2025-01-26:
Hmm. I'll close this, since it is not a common request. Maybe my bit-breaking issues are due to me using fast motion rates, and using glass-epoxy PCBs. 

Anyway, Ill do a few more test runs, and can always do as you suggest. 

Thanks. Nice software! I used to use FlatCAM, but it seems now a pain to run in modern Linux distros.

closed 2025-01-26 by m-anish